davidferguson/pibakery #129: PiBakery SD drive write failed

gerodonovan1, 2018-01-23:
Hi running windows 10 creating Raspberry PI SD using PiBakery failed.
Getting write error on SD card.
When I insert my SD card to PC see 2 devices:
H: Drive labelled SD with 40Mb.
I: Drive with 14.5 GB
When I try to write to SD using PiBakery I only have option to use H: drive.
20MB is written then fails with "SD write failed" error.
Tried the usual reboot reinstall PiBakery.

davidferguson, 2018-01-23:
With the error showing, if you use the shortcut **CTRL**-**Shift**-**I** it will open the Developer Inspector. If you choose the **Console** tab, more detailed errors will be shown. Could you either copy-paste the errors here or just take a screenshot of the errors?

gerodonovan1, 2018-01-23:
Hi David,
Detail from Console:
[image]

Many thanks,

Ger.

davidferguson, 2018-01-23:
I can't seem to see any error in the console, but I'm thinking that the issue is probably because of the way you've partitioned the SD card. The best fix for this is to reformat the SD card using [SD Memory Card Formatter](https://www.sdcard.org/downloads/formatter_4/). Once you've formatted the card try it again with PiBakery.

gerodonovan1, 2018-01-23:
Hi David,
That sorted my problem.
Reformatted SD card and PiBakery loaded correctly.
Much appreciate prompt and helpful response.

Regards,

Ger.

closed 2018-01-23 by gerodonovan1