davidferguson/pibakery #130: Can't write to SD card, because i can't select os

doej1367, 2018-02-14:
Sorry that i'm still running 0.3.4 (don't know if you've fixed this bug already in a newer version), but it would take a week for me to download the newer version.

The problem:
If i click "write", the sd card is selected, but I cant choose any operating system.
Sadly, there are no errors, so i can't even guess what it is.
When i click on "Start Write" nothing happens.

During the Install with the .exe the .img files couldn't be copied, but i've added those manually later. 
Maybe this tells someting about the problem.

davidferguson, 2018-02-14:
>During the Install with the .exe the .img files couldn't be copied, but i've added those manually later.
Maybe this tells something about the problem.

I presume you added these to the C:\Wherever-you-insalled-PiBakery\resources\app\os\ directory. Along with those two .img files the installer also creates a file named `images.json` that tells PiBakery what images are installed, what they're called, what version they are, etc.

I expect the issue to be that PiBakery looks for this file but doesn't find it, and therefore can't find the images.

To fix this you should just be able to create a the `images.json` file in the `os\` directory with the contents from https://raw.githubusercontent.com/davidferguson/pibakery-raspbian/master/images.json.

Hope this helps.

doej1367, 2018-02-14:
well this .json file aready exists (replaced it anyway by yours), but this didn't help.
how should those files in this os folder look like?
right now they are called:
+ raspbian-lite-pibakery-new.img
+ raspbian-pibakery-new.img

davidferguson, 2018-02-14:
Ah, they should just be named

- raspbian-lite-pibakery.img
- raspbian-pibakery.img

So exactly the same, just without the `-new` part.

doej1367, 2018-02-14:
perfect it works!
thanks just renamed them and i can choose the os now.

closed 2018-02-14 by davidferguson