davidferguson/pibakery #132: WiFi Setup failing on FirstBoot

damonkh, 2018-03-11:
I have a VERY simple setup script.
Enable VNC
Set Logged Into Desktop
Setup Wifi
runcommand ifconfig

Here is the output on firstboot.log
`Created symlink /etc/systemd/system/multi-user.target.wants/vncserver-x11-serviced.service → /usr/lib/systemd/system/vncserver-x11-serviced.service.
Desktop logged in
Created symlink /etc/systemd/system/default.target → /lib/systemd/system/graphical.target.
Selected interface 'p2p-dev-wlan0'
OK
mesg: ttyname failed: Inappropriate ioctl for device
mesg: ttyname failed: Inappropriate ioctl for device
Err:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease
  Temporary failure resolving 'mirrordirector.raspbian.org'
Err:2 http://archive.raspberrypi.org/debian stretch InRelease
  Temporary failure resolving 'archive.raspberrypi.org'
Reading package lists...
W: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/stretch/InRelease  Temporary failure resolving 'mirrordirector.raspbian.org'
W: Failed to fetch http://archive.raspberrypi.org/debian/dists/stretch/InRelease  Temporary failure resolving 'archive.raspberrypi.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
mesg: ttyname failed: Inappropriate ioctl for device
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.`

Now, after first boot script is run, the reboot occurs and desktop is logged in AND Wifi is correctly setup.

The problem?  I need WiFi on Firstboot and it is failing.  So, nothing after works.  I have a fairly complex script that is dependent upon upgrade and installing additional software.  All of which works fine if I connect via Ethernet.  In production, I will not have the ability to connect via ethernet.

Really need some help.

Cheers

Barry4679, 2018-03-12:
I see this problem also. I think that it started happening when I installed the new Stretch version.
I use the same workaround: Ethernet

davidferguson, 2018-03-12:
Is this with a Pi3/0W onboard WiFi, or another Pi with a USB adapter?

I'm suspecting a USB adapter (and the rest of this message assumes this assumption is correct), as having researched this now, Stretch now uses different device names for network interfaces, so when the script refers to wlan0, it won't find that and get confused.

Sadly I'm at university right now and university WiFi networks are an utter pain and almost impossible to connect to on a good day from Linux, so I can't test this myself right now, but will try to in the upcoming weeks.

If you log into your Pi and run `ifconfig` with the USB wifi plugged in, can you paste the output here?

Barry4679, 2018-03-12:
In my case (not the OP) I have Pi3 with onboard WiFi.

question what is "WiFi country" in RPi Preferences|Localisation used for? It comes set to GB ... I am in Australia.

WiFi is working now, and was set up correctly by PiBakery .... but as with the OP, it wasn't available during FirstBoot ... causing FirstBoot to fail unless RPi was attached via Ethernet. .... I tried over multiple days, so not related to Internet issues.

ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.7  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::fe65:3616:1beb:aa2a  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:69:c1:ce  txqueuelen 1000  (Ethernet)
        RX packets 33923  bytes 8179370 (7.8 MiB)
        RX errors 0  dropped 1858  overruns 0  frame 0
        TX packets 26903  bytes 15391055 (14.6 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 228  bytes 78788 (76.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 228  bytes 78788 (76.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.21  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::e974:5834:3372:1a3e  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:3c:94:9b  txqueuelen 1000  (Ethernet)
        RX packets 10516  bytes 2266731 (2.1 MiB)
        RX errors 0  dropped 1835  overruns 0  frame 0
        TX packets 1085  bytes 393742 (384.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

davidferguson, 2018-03-12:
@Barry4679 I'd be interested to see the output of your `firstboot.log` file. I think it should still be there in /boot/PiBakery/

Barry4679, 2018-03-12:
The FirstBoot was OK because it was created when Ethernet attached.

I recreated RPi w/o Ethernet.

FirstBoot starts like this

Selected interface 'p2p-dev-wlan0'
OK
Desktop logged in
Created symlink /etc/systemd/system/default.target → /lib/systemd/system/graphical.target.
Err:1 http://archive.raspberrypi.org/debian stretch InRelease
  Temporary failure resolving 'archive.raspberrypi.org'
Err:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
  Temporary failure resolving 'mirrordirector.raspbian.org'
Reading package lists...
W: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/stretch/InRelease  Temporary failure resolving 'mirrordirector.raspbian.org'
W: Failed to fetch http://archive.raspberrypi.org/debian/dists/stretch/InRelease  Temporary failure resolving 'archive.raspberrypi.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists...

With Ethernet it started like this
Selected interface 'p2p-dev-wlan0'
OK
Desktop logged in
Created symlink /etc/systemd/system/default.target → /lib/systemd/system/graphical.target.
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]

In both cases WiFi is OK by 2nd boot.

Here is start of FirstBoot blocks
[image]

davidferguson, 2018-03-12:
Would you be able to write an SD card again, but the time set the contents of /boot/PiBakery/blocks/wifisetup/wifiConnect.py to https://raw.githubusercontent.com/0E800/pibakery-blocks/99207d81cedae96d632b08d1eb32149368cf80c3/wifisetup/wifiConnect.py ?

It's a pull request that's been created for this fix, but I've not been able to test it because of my rather complicated university WiFi setup.

Barry4679, 2018-03-12:
yes I am willing to test

going out for an hour, can do so when I return

But you need to explain ... how to "set the contents of /boot/PiBakery/blocks/wifisetup/wifiConnect.py to ....."

PS: I am a Windows guy ... next to no Linux ... PiBakery is running on my Windows PC.

where is the python program you refer to?

surely not on the RPi ... doesn't all that get blitzed by FirstBoot? ... how do i get control before FirstBoot runs?

Barry4679, 2018-03-12:
is OK I see it on the SD card in Windows.

davidferguson, 2018-03-12:
Apologies, what I mean is edit the file on the Windows machine after it's written in PiBakery, but before you eject the SD card.

So basically:
* Write SD card in PiBakery
* Go into Windows Explorer, and open the SD card drive
* Go into the `PiBakery` folder, then `blocks`, then `wifisetup`
* Open `wifiConnect.py` in any text editor (could be Notepad)
* Copy the contents of https://raw.githubusercontent.com/0E800/pibakery-blocks/99207d81cedae96d632b08d1eb32149368cf80c3/wifisetup/wifiConnect.py and paste it into `wifiConnect.py` overwriting the contents
* Save, and eject the SD card

Barry4679, 2018-03-12:
gotya ... is booting, will get back to you when i return

Barry4679, 2018-03-12:
the extra couple of steps fixed it (for me)

Selected interface 'p2p-dev-wlan0'
OK
Desktop logged in
Created symlink /etc/systemd/system/default.target → /lib/systemd/system/graphical.target.
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease [15.0 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages [11.7 MB]
[image]

davidferguson, 2018-03-12:
Excellent to hear. I had an update planned this week anyway, and I'll include that patch in it.

I'll leave this issue open for now, and close it when that update is pushed out.

Barry4679, 2018-03-12:
thanks for your support

Barry4679, 2018-03-12:
anywhere that I can monitor to see when update is pushed out?

davidferguson, 2018-03-12:
[The PiBakery twitter](https://twitter.com/PiBakery) is the only place right now - I keep meaning to get a blog up and running but never have enough time to sit down and actually do it.

You'll also get a notification in PiBakery the next time you open it after the update is pushed out, asking if you want to update.

tri7, 2018-03-12:
Hi

I can also confirm that this worked for me - recipe wouldn't download a file, nor apt -y upgrade, etc, but after ssh'ing there would be connection. Thanks for the fix.

Btw, in firstboot.log i see "mesg: ttyname failed: Inappropriate ioctl for device" Any ideas?

Thanks again

damonkh, 2018-03-12:
Just checked the bug report.  I will report back after trying the updated .py script.  Thanks everyone for jumping on this!

damonkh, 2018-03-12:
Ok. @davidferguson thanks ever so much for jumping on this.  Your changes to wifiConnect.py fixed the problem.

You have made it possible for me to provide a setup script for my team in Switzerland.   This was a huge issue for us.

You are going to go far in your career!

ashleycawley, 2018-03-12:
I believe I may have been bumping into the same problem: I was using quite a complex PiBakery recipe (writing as Raspbian Lite), deployed to Raspberry Pi 2 Model B V1.1 running on built-in ethernet it worked fine. However the same recipe deployed to a Pi Zero W V1.1 over built-in WiFi would not work. The PiZero was connected to the network however it didn't appear to fullfill most of its recipe, probably because a lot of it was network dependent.

The firstboot.log showed:
Selected interface 'p2p-dev-wlan0'
And then a lot of issues resolving DNS trying to connect to places to download packages, presumably just a connectivity / network issue.

As I say despite the script not completing on the PiZero it was connected automatically to my WiFi (& on a custom SSH port) so I was able to login and manually execute:
cd /boot/PiBakery/ && sudo ./firstBoot.sh
And that did the trick and got me my usual working system. So it seems as though on the initial boot PiBakery isn't connecting to the network over built-in WiFi on the PiZero when it needs it most. I will check the latest updates. Thanks as always @davidferguson for your work on this, I know your life is also busier with more important things!

ashleycawley, 2018-03-12:
I believe I may have been bumping into the same problem: I was using quite a complex PiBakery recipe (writing as Raspbian Lite), deployed to Raspberry Pi 2 Model B V1.1 running on built-in ethernet it worked fine. However the same recipe deployed to a Pi Zero W V1.1 over built-in WiFi would not work. The PiZero was connected to the network however it didn't appear to fullfill most of its recipe, probably because a lot of it was network dependent.

The firstboot.log showed:
Selected interface 'p2p-dev-wlan0'
And then a lot of issues resolving DNS trying to connect to places to download packages, presumably just a connectivity / network issue.

As I say despite the script not completing on the PiZero it was connected automatically to my WiFi (& on a custom SSH port) so I was able to login and manually execute:
cd /boot/PiBakery/ && sudo ./firstBoot.sh
And that did the trick and got me my usual working system. So it seems as though on the initial boot PiBakery isn't connecting to the network over built-in WiFi on the PiZero when it needs it most. I will check the latest updates.

Thanks as always @davidferguson for your work on this, I know your life is also busier with more important things! If I can be of any assistance in testing or debugging then let me know, I have access to practically every model of Pi & method of connectivity.

ashleycawley, 2018-03-12:
I can confirm that modifying the /boot/PiBakery/blocks/wifisetup/wifiConnect.py file on the SD Card to contain this contents instead: https://raw.githubusercontent.com/0E800/pibakery-blocks/99207d81cedae96d632b08d1eb32149368cf80c3/wifisetup/wifiConnect.py fixed the issue for me on my PiZero with built-in WiFi.

@davidferguson Taking a peak at the code before and after I noted that **daemon-reload** is used/restarted and **dhcpcd** is also restarted with 5 second sleeps in between each, is that a summary of the changes or were there more elsewhere that I missed? Thanks for your time and help!

davidferguson, 2018-03-14:
That's the update pushed out, including fixes for:
- `wifisetup` outlined here
- fix for `updateupgrade`
- fixes for `wifisetup` for the new Raspbian version released today as it's now required to set the `country=` parameter in `wpa_supplicant.conf` using `raspi-config`

WiFi on first boot should be working again, so I'm closing this issue, but if any of you run into the same issue just reopen it and I can take a look.

----

>@davidferguson Taking a peak at the code before and after I noted that daemon-reload is used/restarted and dhcpcd is also restarted with 5 second sleeps in between each, is that a summary of the changes or were there more elsewhere that I missed? Thanks for your time and help!

That's exactly it - it's actually not my fix, thank @0E800 for submitting davidferguson/pibakery-blocks/pull/78 which is the fix for it.

----

>Btw, in firstboot.log i see "mesg: ttyname failed: Inappropriate ioctl for device" Any ideas?

@damonkh I get this too, and I'm not sure what's causing it. I'll look into the source and open another issue if necessary, as it's not affecting the WiFi on first boot.

closed 2018-03-14 by davidferguson

Barry4679, 2018-03-15:
I opened PiBakery today, and it did the big download.

I says version 0.3.7 when I open it. ... expected?

I don't see the new code in wifisetup
It ends like this

<<<<<
with open("/etc/wpa_supplicant/wpa_supplicant.conf", "a") as wifiFile:
wifiFile.write(wifiText)

os.system("wpa_cli reconfigure")
time.sleep(5)

# It's likely that the block following this one will be one that uses the
# internet - such as a download file or apt-get block. It takes a few
seconds
# for the WiFi to connect and obtain an IP address, run the waitForNetwork
shell
# script, which will loop waiting for a network connection (timeout 150
seconds)
# and continue once there is one
os.system("chmod +x /boot/PiBakery/blocks/wifisetup/waitForNetwork.sh")
os.system("/boot/PiBakery/blocks/wifisetup/waitForNetwork.sh")

>>>>>

WiFi is not available at the beginning of FirstBoot

Also I can't see any change with respect to WiFi country code configuration.

Did i jump the gun? ... Was there something else I needed to do to get the
new blocks? ... Is the problem at my end or your end?

davidferguson, 2018-03-15:
The way PiBakery does updates isn't brilliant at the moment (will be changed in the next major release).

Currently, it first checks for Raspbian updates, then Raspbian Lite updates, and then finally block updates. However, **it will only do one update per application restart**. Therefore, you actually need to restart PiBakery three times to get all three updates (Raspbian Full, Raspbian Lite, and the blocks).

davidferguson, 2018-03-15:
Update: I've just installed an older version of PiBakery, and gone through all three updates, and the third update (the block update) does indeed include this fix.

Barry4679, 2018-03-15:
I have restarted more than 3 times.

Not getting the new blocks. 

I said I did the long update ... I "think" I did it ... I was expecting it due to email ... I opened PiBakery, and it offered me the download ... I am fairly sure that I agreed .. I clicked something ... the download option I think ... and then went for coffee to kick-start my actual waking up process :} ... so I didn't observe the download ... I believe that I saw it starting.

Is there an install log anywhere? 

I don't get offered the option again when I restart PiBakery ... is it waiting some duration before offering again? ... can I reset it?

I can't see anything with a recent date upon it.

Barry4679, 2018-03-15:
raspbian-pibakery.img has date of 14th March

davidferguson, 2018-03-15:
If you open PiBakery, and then use the shortcut **CTRL** - **Shift** - **I** to open the developer inspector, are there any errors in there?

Can you copy the contents of the file
`C:\Program Files (x86)\PiBakery\resources\app\os\images.json` and also
`C:\Program Files (x86)\PiBakery\resources\app\os\info.json` and paste them here?

Barry4679, 2018-03-15:
I didn't see any obvious error messages in the Console
Just one warning: "'KeyboardEvent.keyIdentifier' is deprecated and will be removed in M53, around September 2016. See https://www.chromestatus.com/features/5316065118650368 for more details."

which part of the UI should I look in?

I don't see any info.json file in that directory

images.json
[{"version":7,"compressedMD5":"cb29576520ed28fd8f0d8468c75f3005","uncompressedMD5":"17456aeb6568e0381c2443a44eea925b","downloadUrl":"https://github.com/davidferguson/pibakery-raspbian/releases/download/v0.6.0/raspbian-pibakery.7z","uncompressedSize":4919918592,"filename":"raspbian-pibakery.img","compressedFilename":"raspbian-pibakery.7z","uncompressedFilename":"raspbian-pibakery-new.img","displayName":"Raspbian Full","installed":true,"skipVersion":0},{"version":6,"compressedMD5":"81754f93de945926c93bab3f85ef5b2e","uncompressedMD5":"6300db151cf65f4f55a4f4c9bc1e2840","downloadUrl":"https://github.com/davidferguson/pibakery-raspbian/releases/download/v0.6.0/raspbian-lite-pibakery.7z","uncompressedSize":1858076672,"filename":"raspbian-lite-pibakery.img","compressedFilename":"raspbian-lite-pibakery.7z","uncompressedFilename":"raspbian-lite-pibakery-new.img","displayName":"Raspbian Lite","installed":false,"skipVersion":0}]
=======================

Barry4679, 2018-03-15:
I can't see any failure in the Windows event log.

maybe something strange, and local, happened during the upgrade.

Is there somewhere where I can download the prior version, so that I can then closely observe the upgrade?

davidferguson, 2018-03-15:
So from that I can see that the **Raspbian Full** update (the first one) completed fine, but it's not attempted to do the second **Raspbian Lite**, and therefore not the blocks update either.

It's very possible this is a caching error, with some server somewhere storing and serving an older version of the update json, which is preventing PiBakery from seeing the update.

The easiest fix for this is to simply download the latest full PiBakery application, which contains the latest Raspbian Full, Lite and blocks. The installers are available at http://www.PiBakery.org/download.html

Barry4679, 2018-03-15:
I am not in any urgent hurry.
I can wait a day to verify your cache theory if you like. ... ?

davidferguson, 2018-03-15:
There's actually a faster way to check what updates PiBakery is 'seeing'.

If you open up the developer inspector again, and then type in the following and hit enter:

```
request.get('https://raw.githubusercontent.com/davidferguson/pibakery-raspbian/master/images.json', function (error, response, body) {
  console.log(error, body)
})
```

Unfortunately there's an annoying bug in PiBakery that means you can't paste in the developer inspector, so you'll have to type it all out manually! As pressing enter runs the code you've entered, to get a new line you need to use shift+enter.

This should print out a whole bunch of JSON, if you could paste that here that would be very useful.

Barry4679, 2018-03-15:
<<<
Unfortunately there's an annoying bug in PiBakery that means you can't paste in the developer inspector, so you'll have to type it all out manually!
>>>

I am sceptical by nature ... cut and paste worked for me ... Chrome, Win10
 
request.get('https://raw.githubusercontent.com/davidferguson/pibakery-raspbian/master/images.json', function (error, response, body) {
  console.log(error, body)
})
Request {domain: null, _events: EventHandlers, _eventsCount: 3, _maxListeners: undefined, uri: Url…}
VM688:2 null "[
	{
		"version": 7.0,
		"compressedMD5": "9756d47750543ab0c75182fd8e5c42bd",
		"uncompressedMD5": "1747e828ba9bec478175051675012f54",
		"downloadUrl": "https://github.com/davidferguson/pibakery-raspbian/releases/download/v0.7.0/raspbian-pibakery.7z",
		"uncompressedSize": 4949278720,
		"filename": "raspbian-pibakery.img",
		"compressedFilename": "raspbian-pibakery.7z",
		"uncompressedFilename": "raspbian-pibakery-new.img",
		"displayName": "Raspbian Full",
		"installed": true,
		"skipVersion": 0
	},
	{
		"version": 7.0,
		"compressedMD5": "91b788d00de808a03bc83b6983ea3f2d",
		"uncompressedMD5": "a651e4c5eba71bdf566ea7555f2b2684",
		"downloadUrl": "https://github.com/davidferguson/pibakery-raspbian/releases/download/v0.7.0/raspbian-lite-pibakery.7z",
		"uncompressedSize": 1858076672,
		"filename": "raspbian-lite-pibakery.img",
		"compressedFilename": "raspbian-lite-pibakery.7z",
		"uncompressedFilename": "raspbian-lite-pibakery-new.img",
		"displayName": "Raspbian Lite",
		"installed": true,
		"skipVersion": 0
	}
]
"

Barry4679, 2018-03-19:
Is there any update upon this issue?

I know that I can delete and then re-download PiBakery ... but in the meantime it seems to be stuck in a half updated state? Do you need any more info about this?

Barry4679, 2018-03-23:
see two further posts to this issue at
https://github.com/davidferguson/pibakery/issues/132
they never made it to this email thread for some reason

davidferguson, 2018-03-23:
Sorry, I must have missed the last few comments on this thread.

It's very very odd, and an issue I've never come across before. The easiest fix is going to be to do a re-install, but if you want to help debug this, then:

1. Open PiBakery
2. When/if asked if you want to update, say **Remind me later**
3. Open the developer inspector with CTRL-Shift-I
4. Paste in the contents of https://gist.github.com/davidferguson/dcd8ccc39b5c700a404241b72029fd6f
5. Then paste in:
```
checkForRaspbianUpdates(function () {
  checkForBlockUpdates()
})
```

And then copy-paste the output of that.

Barry4679, 2018-03-23:
not given an update option

checkForRaspbianUpdates(function () {
  checkForBlockUpdates()
})
undefined
VM490:8 jsondata: [Object, Object]
VM490:18 installedImages: ["raspbian-pibakery.img"]
VM490:19 installedImagesIndexes: [0]
VM490:29 github images: [Object, Object]
VM490:42 i: 0
VM490:43 if1: false
VM490:44 if2: true

davidferguson, 2018-03-23:
Ah, found the bug. It'll be fixed in the next version of PiBakery. Thanks very much for being so helpful with this!

The easiest solution is just to re-install - but thanks again, as I don't think I would have ever spotted this bug on my own!

I've moved it to a separate bug as it has nothing to do with this bug (WiFi setup failing) - so see #135 for more information and updates.

Barry4679, 2018-03-23:
Good, Thanks for keeping me in the loop.