davidferguson/pibakery #134: This build error occurs from a fresh source download - posted March 27

BearChavez, 2018-03-21:
> PiBakery@0.3.8 start /home/bear/Desktop/pibakery
> electron .

sh: 1: electron: not found

npm ERR! PiBakery@0.3.8 start: `electron .`
npm ERR! Exit status 127
npm ERR! 
npm ERR! Failed at the PiBakery@0.3.8 start script.
npm ERR! This is most likely a problem with the PiBakery package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron .
npm ERR! You can get their info via:
npm ERR!     npm owner ls PiBakery
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 4.14.0-parrot13-686-pae
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "start"
npm ERR! cwd /home/bear/Desktop/pibakery
npm ERR! node -v v8.9.3
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/bear/Desktop/pibakery/npm-debug.log
npm ERR! not ok code 0

I will post the debug log if needed

davidferguson, 2018-03-21:
Did you follow the instructions listed at https://github.com/davidferguson/pibakery#installing-from-source?

From the output you've posted it looks like you haven't run `npm install`, as that *should* install electron, which is why it isn't running at the moment.

BearChavez, 2018-03-21:
@davidferguson I'll run it again and post the results after trying start