davidferguson/pibakery #141: Remove PiBackery

mariotti, 2018-04-28:
It would be nice to have a task called: remove pibakery. Reasons:

 - Just because: I want MY sd card back.
 - Security: I do not need extra hooks

Maybe write to /etc/motd, if it exists anymore, "created with pibakery".

ashleycawley, 2018-05-09:
This doesn't make sense. I'm assuming your point may have been lost in translation.

mariotti, 2018-05-12:
Ok lets assume I want to distribute my SD card/my settings for a given application.
An example: a security camera setup.

What I would like to have is an SD card which once booted does:
 - Use pibackery to make the set up.
 - reboot with my needed configs.
 - remove pibackery.

This because, once it has been set up, I do not expect the final user to hack it again.

The pibackery files can be removed, and make sure we do not have extra hooks.

As I was mentioning we can keep trace of the process. Like "made with pybackery".

It is a kind of: this SD is final. 

I am not sure why this should not make sense. Of course users can hack the SD and the raspy,
but I do not want to make it available as a standard process.

I do not want, that pybackery is the the way to hack the system just because.

ashleycawley, 2018-05-13:
Thanks for the additional information. Just to clarify it is spelt PiBakery.

> make sure we do not have extra hooks

I think this is the main area of your concern. But are there any "hooks" left after the fact? Yes I know there is a PiBakery folder left over which contains logs and first-run scripts etc. but are there any services, hooks or other **running** remanent? I don't know myself, developers of the project might be able to best advise us on that point. But from what I've seen I haven't seen any running left-overs which get in the way, just static files.

If that was the case and lets say all left-over files were contained in /boot/PiBakery/ then you may be able to simply remove that directory (and you might be able to use PiBakery to do that for you)? I'm not sure if that is safe or whether the system is relying on it going-forward, perhaps someone else can enlighten us on that? I may do some testing soon if I get time.

punggolzenith, 2019-01-03:
Yes i second this, because if we set root passwords, or send certificates with PiBakery, the credentials will be left there.

I tried run with `root` `rm -rf /boot/PiBakery/ `  I will let you know if it works tomorrow.

ashleycawley, 2019-01-04:
> Yes i second this, because if we set root passwords, or send certificates with PiBakery, the credentials will be left there.

Not necessarily, I don't know for sure because I haven't sat down to analyse the code yet and how it works, _but_ it isn't infeasible for PiBakery to set passwords on the local system and then erase the password so it is not stored anywhere within the PiBakery files or logs, in fact I would have thought that would be the logical approach to follow.

You can probably test this theory by grep'ing for your password in the /boot/PiBakery/ directory, have you tried that? I might take a look myself if I get a chance soon.

ashleycawley, 2019-01-04:
I was quite surprised but I did find the password saved on record, if you have used PiBakery to change a password for you on first boot, give this command a go on the Raspberry Pi:
`grep -ri "changepass" /boot/PiBakery/firstBoot.sh`
And you will likely see your password somewhere within the results (I did).

I must admit seeing that has confused me a little because when inspecting the file at:
/boot/PiBakery/blocks/changepass/changepass.py
or the code at:
https://github.com/davidferguson/pibakery/blob/master/pibakery-blocks/changepass/changepass.py

It looked like to my naive eyes that the code was going to some lengths to either hide or mask the password perhaps. I'm new to coding so I cannot fully interpret it unfortunately.

I'm still new to the idea of contributing but if I knew @davidferguson was behind the idea I would be happy to help by trying to code some routine to do some post clean-up or sanitization of passwords etc.

Is that worth me pursuing @davidferguson or is it not something that would be feasible for technical reasons unknown to me atm?