davidferguson/pibakery #142: Wifi is not started when using AT

wilfredkl, 2018-05-02:
Wifi does noet work if you enter country code AT.
Error in log stating that country code is not set.
works when changed to GB.
checked the iso standard just to be sure this is the correct country code and it is.
havent tried other countries but may need checking aswell.

[image]

davidferguson, 2018-08-02:
@exocode I think this is the first reported success of someone using an image other than Raspbian or Raspbian Lite!

@wilfredkl Hmm, that's odd. The block just sets it in the wpa_supplicant.conf file, so I'm not sure why it doesn't work. I'll ask Raspberry Pi if there's any issues with country codes next week.

davidferguson, 2018-08-31:
The country codes are forced to be required on a Pi 3+, and the newer Stretch versions (Feb 2018 and beyond). I'm guessing you were using Raspbian from then or onwards, @wilfredkl?

I'll investigate with my Pi 3+, and see if I can solve this.