davidferguson/pibakery #144: Bundling PiBakery into a standalone installer

maskaravivek, 2018-05-23:
I have made a few changes in pibakery to make it compatible with my usecase and am using it by running `npm start` every time i need to use it. 

I was wondering how I could bundle everything into a installer so that it can be distributed to other computers. Having the users follow the installation steps would be too much to ask for. 

@davidferguson It would be great if you could give insights about how everything(pibakery along with the modified raspbian images) are bundled together to an executable.

clayreimann, 2018-08-10:
You're looking for these scripts: https://github.com/davidferguson/pibakery/blob/master/package.json#L10-L12

davidferguson, 2018-08-13:
Yep, those are what makes the executables. To make the installers, I use [Packages](http://s.sudre.free.fr/Software/Packages/about.html) for Mac, and [InnoSetup](http://www.jrsoftware.org/isinfo.php) for Windows, although I'll probably be changing to a standard .dmg "installer" for Mac soon.

closed 2018-08-13 by davidferguson

maskaravivek, 2018-08-13:
Thanks, guys for the information. Just to clarify, are you able to package the OS image as well? The packaging fails for me if I try to bundle it.

gurtajs4, 2019-08-05:
Hi @davidferguson 
Can you please share the Inno Setup .iss file used to create EXE installer for windows
Appreciated your help 

Regards
Gurtaj