davidferguson/pibakery #147: Feature Request: Option to download images and de-compress

gauntface, 2018-06-21:
(On linux at least) the images can be defined but if the img doesn't exist, no warning, errors or information is displayed to help diagnose the problem.

It would be awesome if the app would provide an option to download known images in images.json as well show information of what the app can find and use outside of the the write drop down.

I'm happy to flesh out a proposal and work on it.

davidferguson, 2018-06-22:
PiBakery is actually in the process of being completly re-written (which is why I've been slow to accept PR and answer issues), with release due to be in the next few days/this week.

This update will do away with including the images with PiBakery, and instead let the user download and choose whatever (Raspbian) image they want to use.

I'll leave this open until the new version is available.

gauntface, 2018-06-23:
@davidferguson :heart: this response a lot and doubly glad I asked first ;)

Let me know if you have any low hanging fruit I can help with.

davidferguson, 2018-06-29:
> I'll leave this open until the new version is available.

Now released.

> Let me know if you have any low hanging fruit I can help with.

Bug finding and fixing with PiBakery v2 at the moment, basically! It's an entirely new codebase, so there will be lots of fresh bugs to squash.

closed 2018-06-29 by davidferguson