davidferguson/pibakery #148: Feature Request Change config without reflashing

DragonHunter274, 2018-06-21:
I don't know if this is possible but can you add  the ability to change for example the wifi credentials without reflashing the sd card?

davidferguson, 2018-06-22:
In response to initial question: Currently it's possible for SD cards that were written with PiBakery. However in the next few days a major update will be released that will support editing SD cards that weren't written in PiBakery.

----

>EDIT 2: I just tried to use the update button in the ui but it doesn't react to being clicked on

>EDIT 3: apparantly ther is a bug if you have multiple storage devices connected it tries to write to the wrong one in the console it then says blocks.xml not found (which it can't find it is searching on the wrong drive)

Hmm, interesting. I probably won't fix this for now, as the new update mentioned above edits SD cards in a different way, and therefore this should no longer be an issue.

I'll leave this open until the new version is available.

davidferguson, 2018-06-29:
> However in the next few days a major update will be released that will support editing SD cards that weren't written in PiBakery.

Now been released (although still a pre-release - **there will be bugs!**)

closed 2018-06-29 by davidferguson