davidferguson/pibakery #156: SD Card drop down stays at "loading..."

dgriswo, 2018-07-09:
When selecting Load, Update, or Write, the pull down for SD Card: remains at "loading..." and does not populate with available SD card.  

PiBakery 2.0.0 on Windows 10 version 1709.
SD card is a Samsung EVO Plus in a Canakit microSD reader.

In an attempt to isolate the issue....
1. Ran PiBakery.exe as a normal user and as an administrator
2. Inserted the SD Reader before and after PiBakery launch.
3. Used a SD card with Raspian installed, completely blank (no partitions), and FAT32 formatted.
4. SD Card/Reader works on Linux version of PiBakery
5. SD Card/Reader works on Windows using etcher.io to write raspbian .img

davidferguson, 2018-07-09:
If you use the shortcut CTRL-Shift-I this will open the Developer Inspector. Paste in the following code and hit enter, and tell me what you get as a result:

```
var drivelist = require('drivelist')
drivelist.list(function (error, drives) {
  console.log('error:', error)
  console.log('drives:', drives)
})```

dgriswo, 2018-07-09:
var drivelist = require('drivelist')
drivelist.list(function (error, drives) {
  console.log('error:', error)
  console.log('drives:', drives)
})
D:\Users\Downloads\PiBakery-win32-ia32\resources\app\node_modules\bindings\bindings.js:88 Uncaught Error: \\?\D:\Users\Downloads\PiBakery-win32-ia32\resources\app\node_modules\drivelist\build\Release\drivelist.node is not a valid Win32 application\\?\D:\Users\Downloads\PiBakery-win32-ia32\resources\app\node_modules\drivelist\build\Release\drivelist.node
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:172:20)
    at Object.Module._extensions..node (module.js:671:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:172:20)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (module.js:586:17)
    at require (internal/module.js:11:18)
    at bindings (D:\Users\Downloads\PiBakery-win32-ia32\resources\app\node_modules\bindings\bindings.js:81:44)
    at Object.exports.list (D:\Users\Downloads\PiBakery-win32-ia32\resources\app\node_modules\drivelist\lib\drivelist.js:53:5)
    at <anonymous>:2:11

davidferguson, 2018-07-10:
Hmm, not sure what's causing that. I'll need to investigate on my Windows computer.

davidferguson, 2018-07-10:
I'm able to reproduce on a Windows 10 computer. Will investigate.

davidferguson, 2018-07-11:
It seems that producing one 32 bit version of PiBakery is no longer sufficient, and I'll need to do both ia32 and x64 from now on, as the ia32 version doesn't run on Windows 10 64 bit.

I've built a x64 version and uploaded it on the [v2.0.0 Release Page](https://github.com/davidferguson/pibakery/releases/tag/v2.0.0) - can you try this and confirm it works?

closed 2018-07-11 by davidferguson

davidferguson, 2018-08-30:
Using the x64 build is a conformed fix for this.

commit a78d0b02 (2018-08-30): make build-win build for 64 bit

fixes #156