davidferguson/pibakery #158: Support for other images than Raspian

exocode, 2018-07-25:
Hi there, 

I just started with RPI and PiBakery.

Are you planning to support other images as well? 
In my case I am interested in:
- jessie minibian (https://minibianpi.wordpress.com/download/)
- hypriotos (https://blog.hypriot.com/downloads/)

Or is this technically not possible..?

davidferguson, 2018-07-25:
It's quite a big challenge technically. PiBakery v2 (which is still in pre-release but can be downloaded if you're brave!) supports chosing your own image, but this will only work for Debian derivatives, such as Raspbian, Raspbian Lite, etc. I've not tested anything else, but Minibian *might* work - but probably not with all the blocks.

If you want to try it with a different OS and report back here, please do!

I'll be closing this for now because it's not something that I'm planning on implementing.

closed 2018-07-25 by davidferguson

exocode, 2018-08-01:
I am now tried to install the hypriotOS, which is a Raspbian Stretch with Linux Kernel version 4.9.x, which does nothing else than provide a on the go "Docker" image with all it's benefits. 

What I found out is, that hypriotOS is using cloud-init / cloud-config to setup everything. 
I opened a feature request in your repo here: https://github.com/davidferguson/pibakery/issues/161