davidferguson/pibakery #160: Copy / Paste in Blocks (2.x)

exocode, 2018-07-28:
In version 1.x I was able to copy/paste content from outside PiBakery into a field (for example Network and Network Password). This is in version 2 not possible anymore. I am only able write everything by hand, which is a bit annoying, because I use long Wifi passwords...

 [image]

exocode, 2018-07-28:
For now: I found out that drag-and-drop from another app is (still) working. I use a mac.

- select the field you wanna drag in PiBakery
- select your text in a Texteditor
- drag and drop the text into the field

davidferguson, 2018-08-02:
That's odd, it should work in PiBakery v2. v2 uses a different (better) copy-paste system. Do you have options to copy-paste from the Edit menubar?

exocode, 2018-08-02:
The menu bar does not even exist on Mac. Only when I recompile the whole app with `electron .`.
It also is not present when I compile with `electron-packager .`

davidferguson, 2018-08-31:
A very silly bug - I forgot to run the createMenu() command which creates the menu (and enables copy-pasting).

commit b0c14041 (2018-08-31): add in createMenu() to support copying and pasting

fixes #160 #108
closed 2018-08-31 by davidferguson