davidferguson/pibakery #161: [FEATURE] cloud-init support

exocode, 2018-08-01:
It would be nice when V2 will implement/provide the widely used `cloud-init`tool. Maybe in form of a block, where the User can choose a customized "cloud-config.yaml"

Something like this:

```
#cloud-config
## run with --config
# vim: syntax=yaml
#
hostname: black-pearl
manage_etc_hosts: true

wifi:
  interfaces:
    wlan0:
      ssid: "UPC2463EF9"
      password: "zxhWsVb7j3cc"

users:
  - name: pirate
    gecos: "Hypriot Pirate"
    sudo: ALL=(ALL) NOPASSWD:ALL
    shell: /bin/bash
    groups: users,docker,video
    plain_text_passwd: hypriot
    lock_passwd: false
    ssh_pwauth: true
    chpasswd: { expire: false }

package_upgrade: false

# # Set the locale of the system
locale: "de_AT.UTF-8"

# # Set the timezone
# # Value of 'timezone' must exist in /usr/share/zoneinfo
timezone: "Europe/Vienna"

# # Update apt packages on first boot
# package_update: true
# package_upgrade: trueloca
# package_reboot_if_required: true
package_upgrade: false

# # Install any additional apt packages you need here
packages:
  - ntp

# # WiFi connect to HotSpot
# To make wifi work with RPi3 and RPi0
# you also have to set "enable_uart=0" in config.txt
# See no-uart-config.txt for an example.
#
# # - use `wpa_passphrase SSID PASSWORD` to encrypt the psk
write_files:
  - content: |
      allow-hotplug wlan0
      iface wlan0 inet dhcp
      wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
      iface default inet dhcp
    path: /etc/network/interfaces.d/wlan0
  - content: |
      country=YourContryCode
      ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
      update_config=1
      network={
      ssid="MyNetworkName"
      psk="MyPassword"
      proto=RSN
      key_mgmt=WPA-PSK
      pairwise=CCMP
      auth_alg=OPEN
      }
    path: /etc/wpa_supplicant/wpa_supplicant.conf
# These commands will be ran once on first boot only
runcmd:
  # Pickup the hostname changes
  - 'systemctl restart avahi-daemon'
  # Activate WiFi interface
  - 'ifup wlan0'

bootcmd:
 - [ ifup, wlan0 ]
```

I now try to figure out how to install it and will report!

davidferguson, 2018-08-02:
Interesting idea, I've actually never heard of cloud-init before, but its goal seem to overlap significantly with PiBakery so it would be great to integrate the two.

I agree that the most ideal outcome would be to have a PiBakery block that can run a cloud-init script, however this may need a few fundamental changes to PiBakery, as currently you can't attach a file to a block and have that "uploaded" to the Pi; you currently have to move the file manually to /boot (or use the download-file block) and then reference the file's path in the cloud-init block. This is a highly requested feature, so I'll see if it can be added into v2 before it becomes non-beta.

exocode, 2018-08-02:
Sounds great! Yes, I just started digging into the PiBakery code and found out, that it's not possible to copy a local file into the SD. I am not familiar with `Electron`, so I am not a helpful source in that case. 

**Last question:** Why does exist a duplicate (redundant) `pibakery-blocks´ folder within PiBakery when I also fetches the `pibakery-blocks`-Repo? And is the "sort" option really necessary? I am asking, because a lot of people wanna develop their own blocks by forking the repo and wanna test them before starting a PR.

So my idea is, to provide a checkbox and textfield within PiBakery where developers can decide to use a local path or their Github username like in the screenshot below. A simple API call to the regarding github repo results in a list of possible Versions.  (https://api.github.com/repos/davidferguson/pibakery-blocks/releases). So a version can be specified to work with.

[image]

exocode, 2018-10-19:
Only wanna know if you had time to make progress on V2?

davidferguson, 2019-08-11:
> it should be released soon

Well that didn't really work out! Sorry about that, and have a read of https://pibakery.org/future.html for more info on the future of PiBakery.

If I do go forward and re-write a majority of PiBakery for v3, then I'd love to include cloud-init support. I've never used it and am not really familiar with it, so if you want to jump in on #180 and discuss the best ways of doing that, that would be awesome.