davidferguson/pibakery #162: Cant find sd

rhthaakkar, 2018-08-09:
I have tried with bith win10 and win8.. It goes till end and then shows error "can't find sd" 
Pibakery 0.3.8 and Nitro 8GB sd card
[image]

davidferguson, 2018-08-31:
This error indicates that the SD card didn't automatically re-mount once writing was finished.

If you write it again:
1) do you get the same error?
2) if you go to `Windows Explorer -> Computer`, is the SD card listed there? (probably as `boot`)

rhthaakkar, 2018-08-31:
Yeah.. I tried to write again and card was also showing in explorer after
remount but pibakery couldn't detect it.. Changed the card and then
worked..

davidferguson, 2018-08-31:
Odd. SD cards can be weird sometimes.

closed 2018-08-31 by davidferguson