davidferguson/pibakery #165: Pi gets stuck on first boot when testing my block

rick1924, 2018-08-31:
Hi there, I am making a custom block to connect automatically to the network of my university. It is a complex authentication so more lines have to be added to the wpa_supplicant.conf file. I am able to import my block into PiBakery and successfully write to the SD card. However, when I try to boot my Pi, it gets stuck trying to run my block. It stays like this for five minutes and then turns off.

My code is very similar to the original so I really can't tell what I am doing wrong. Below are my scripts:

--------------------------------------
confUbcsecure.py:

#!/usr/bin/python
import sys, os, time
ubcsecure = """

network={
    ssid="ubcsecure"
    scan_ssid=1
    key_mgmt=WPA-EAP
    proto=WPA2 WPA
    eap=PEAP
    group=CCMP TKIP
    pairwise=CCMP TKIP
    ca_cert="/etc/ssl/certs/DigiCert_Global_Root_CA.pem"
    identity=""
    password=""
    phase1="peaplabel=0"
    phase2="auth=MSCHAPV2"
}
"""
countryCode = sys.argv[1]
os.system('raspi-config noint do_wifi_country "' + countryCode + '"')
with open("/etc/wpa_supplicant/wpa_supplicant.conf", "a") as wifiFile:
    wifiFile.write(ubcsecure)
os.system("wpa_cli reconfigure")
time.sleep(5)
os.system("chmod +x /boot/PiBakery/blocks/rfidwifiblock/waitForNetwork.sh")
os.system("/boot/PiBakery/blocks/rfidwifiblock/waitForNetwork.sh")  

-------------------------
rfidwificlock.json:

{
  "name": "rfidwifiblock",
  "text": "Configure Ubcsecure Network\\nCountry %1",
  "script": "confUbcsecure.py",
  "args": [
    {
      "type": "text",
      "default": "CA",
      "maxLength":0
    }
  ],
  "network": false,
  "continue": true,
  "type": "network",
  "category": "network",
  "supportedOperatingSystems": [
    "raspbian-pibakery.img",
    "raspbian-lite-pibakery.img"
  ],
  "shortDescription": "Automatically connects to ubcsecure",
  "longDescription": "This block automatically allows you to connect to the UBC student network"
}

davidferguson, 2018-08-31:
First of all, are you using PiBakery 0.x or PiBakery v2?

When you boot the Pi, what comes on the screen? Does the Pi boot, then go into the "PiBakery installer" (the blue screen with the progress bar)?

And it then hangs on your block, and then shuts down?

rick1924, 2018-08-31:
Thank you for the quick reply. I am using PiBakery 0.x. Yes it does go into the PiBakery installer, for testing purposes I only placed the "On First Boot" and my block, so the progress bar gets stuck at 50%.

I am using a Raspberry Pi 3 Model 3 B+.

davidferguson, 2018-08-31:
It's not unusual for it to take a minute or two for WiFi setup blocks, as it can take a few moments to establish a connection.

Once the block is run, have a look at the contents of wpa_supplicant.conf and see if the settings have been set correctly in there.

Then restart, and see if it connects to the network after the reboot.

rick1924, 2018-08-31:
Just a quick correction on the code I placed here. I mistakenly erased the indentation of the line "wifiFile.write(ubcsecure)".

I have looked at the supplicant file after rebooting the Pi after it crashes and nothing was placed in the file. I have also tried commenting out the lines that wait for the network to be established, but it also gets stuck for the same amount of time. Without the "waitForNetwork.sh" lines, your original code takes about 3s, while mine still times out.