davidferguson/pibakery #166: Pi gets Kernel panic

dopyoman, 2018-09-03:
Hi,

I used PiBakery V2 with a custom raspbian image from Donkey Car [https://docs.donkeycar.com/guide/install_software/#get-the-raspberry-pi-working](url) and even without any configuration blocks just having PiBakery write the image to the SD card fails. I get an 
```
---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00007f00
```

I've tried the image without pibaker and it works.

davidferguson, 2018-09-03:
This is due to a permissions issue with the PiBakery boot scripts. I presume you're running Windows? Did you clone the repository and build from source, or use the downloads from https://github.com/davidferguson/pibakery/releases/tag/v2.0.0 ?

I'll try and publish a new release with this fixed soon.

dopyoman, 2018-09-03:
Ya I'm running windows and used downloaded the release.

johny-mnemonic, 2018-09-14:
Same issue here with PiBakery 2.0 downloaded as binary from release tag.
Everything seems to work when editing SD card, but then Pi ends with Kernel panic...

Is there a manual workaround till it is fixed?

davidferguson, 2018-09-14:
A manual workaround would be to download the zip of the repo, and replace the `resources` folder in PiBakery with the `resources` folder from the downloaded zip.

Or manually set the line endings for the files in `resources` directory to be LF.

Barokai, 2018-11-05:
> A manual workaround would be to download the zip of the repo, and replace the `resources` folder in PiBakery with the `resources` folder from the downloaded zip.

Just tried this, still getting kernel panic :(

tvories, 2019-11-18:
I am also having this issue.  I exported the recipe and built it on my Mac and I do not have the Kernel panic issue.

Is there any way we can get the line endings pushed to a release?