davidferguson/pibakery #167: Fedora aarch64

Blindripper, 2018-09-19:
Is a support of that Distro planned?
What is to do, to support other distros?

BTW Cool project !

Blindripper, 2018-09-19:
Forgot to mention that a Tezos Core Dev recommend this distro.

https://gitlab.com/tezos/tezos/issues/271#note_91200988

davidferguson, 2018-10-14:
Supporting other distros is tricky, as it requires that all the block scripts be written to be compatible with both Debian based distros (as they are currently) and also other distros.

For the moment I'm not officially supporting this, but if people were willing to go and rewrite the [PiBakery block scripts](https://github.com/davidferguson/pibakery-blocks) to be compatible with other distros and submit pull requests, I'd be more than willing to accept them.

closed 2018-10-14 by davidferguson