davidferguson/pibakery #170: electron .

RussNelson, 2018-12-15:
npm ERR! Linux 4.15.0-42-generic
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "start"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! PiBakery@2.0.0 start: `electron .`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the PiBakery@2.0.0 start script 'electron .'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the PiBakery package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron .
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs PiBakery
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls PiBakery
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/nelson/pibakery/npm-debug.log

RussNelson, 2018-12-15:
This is the npm-debug.log

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@3.5.2
3 info using node@v8.10.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle PiBakery@2.0.0~prestart: PiBakery@2.0.0
6 silly lifecycle PiBakery@2.0.0~prestart: no script for prestart, continuing
7 info lifecycle PiBakery@2.0.0~start: PiBakery@2.0.0
8 verbose lifecycle PiBakery@2.0.0~start: unsafe-perm in lifecycle true
9 verbose lifecycle PiBakery@2.0.0~start: PATH: /usr/share/npm/bin/node-gyp-bin:/home/nelson/pibakery/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
10 verbose lifecycle PiBakery@2.0.0~start: CWD: /home/nelson/pibakery
11 silly lifecycle PiBakery@2.0.0~start: Args: [ '-c', 'electron .' ]
12 info lifecycle PiBakery@2.0.0~start: Failed to exec start script
13 verbose stack Error: PiBakery@2.0.0 start: `electron .`
13 verbose stack spawn ENOENT
13 verbose stack     at ChildProcess.<anonymous> (/usr/share/npm/lib/utils/spawn.js:17:16)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at ChildProcess.emit (events.js:214:7)
13 verbose stack     at maybeClose (internal/child_process.js:925:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
14 verbose pkgid PiBakery@2.0.0
15 verbose cwd /home/nelson/pibakery
16 error Linux 4.15.0-42-generic
17 error argv "/usr/bin/node" "/usr/bin/npm" "start"
18 error node v8.10.0
19 error npm  v3.5.2
20 error file sh
21 error code ELIFECYCLE
22 error errno ENOENT
23 error syscall spawn
24 error PiBakery@2.0.0 start: `electron .`
24 error spawn ENOENT
25 error Failed at the PiBakery@2.0.0 start script 'electron .'.
25 error Make sure you have the latest version of node.js and npm installed.
25 error If you do, this is most likely a problem with the PiBakery package,
25 error not with npm itself.
25 error Tell the author that this fails on your system:
25 error     electron .
25 error You can get information on how to open an issue for this project with:
25 error     npm bugs PiBakery
25 error Or if that isn't available, you can get their info via:
25 error     npm owner ls PiBakery
25 error There is likely additional logging output above.
26 verbose exit [ 1, true ]

RussNelson, 2018-12-15:
`npm install` seemed to work just fine but reported these three lines:

npm WARN optional Skipping failed optional dependency //elevator:
npm WARN notsup Not compatible with your operating system or architecture: elevator@2.2.3
npm WARN PiBakery@2.0.0 No repository field.

davidferguson, 2018-12-15:
Hmm odd, it looks almost like an electron issue. What OS are you running? What happens if you run the command ``electron .`` in your prompt?

RussNelson, 2018-12-17:
Turns out to be a missing dependency, libgconf-2.4. `apt-get install libgconf-2.4` resolved the issue. I'm guessing that something you're using already installed that library and so you didn't notice the missing dependency. I won't close this issue. I'll let you close it after you fix the missing dependency or report it to the electron folks.

chris007de, 2019-01-03:
Having the same/a similar issue. `npm start` yields the same error as for @RussNelson, but libgconf-2.4 is already installed. Though, `electron .` starts pibakery just fine.
(Linux x64, Debian Stretch with Kernel 4.19)

lurch, 2019-02-13:
The missing libgconf sounds similar to https://github.com/balena-io/etcher/issues/1305