davidferguson/pibakery #172: First boot wifi issue with raspbian strecht ?

romaindudek, 2019-01-21:
Hi,

First of all, thanks a lot for this wonderfull app ! Great idea and great dev !

I'm facing a little issue wich I think might be related to latest images of debian (strecht)... I need my pi to connect via wifi at the fisrt boot as I need some downloads to occur at first boot to complete my configuration process...

And I dont think it works properly... I have to wait for about 4 minutes10sec before the wifi process is complete (wich I think is the duration of the loop in the waitForNetwork.sh script).

And I wonder about the
os.system("wpa_cli reconfigure")
instruction in the wifiConnect.py file...

I think (maybe) it is not enough to relaunch the wifi daemon...

Do you have any idea ?
Maybe adding 
sudo ifdown wlan0
sudo ifup wlan0
would work ?

Thanks !

oliveiraallex, 2019-02-21:
I fixed this issue in this PR https://github.com/davidferguson/pibakery/pull/174 
You can fix in your SD card editing the same line file wifiConnect.py before insert in the Raspberry.
What is happening, is the command wpa_cli reconfigure is selecting another interface (p2p-dev-wlan0) and not the wlan0. So you need to set this parameter "-i wlan0" to its work.

oliveiraallex, 2019-03-05:
While the fix is not merged, you can also create a block "Run Command" on First Boot with the command below. It will create the correct line in the file. This block needs to be the first block. 

ex -s -c '49i|os.system("wpa_cli -i wlan0 reconfigure")' -c x /boot/PiBakery/blocks/wifisetup/wifiConnect.py