davidferguson/pibakery #175: Raspberry Pi 3 A+ Support

Plasmarobo, 2019-03-08:
It looks like the raspbian image might be out of date for 3 A+ devices.
I have an image that boots on a 3 B+.
It configures WiFi, installs avahi-daemon, changes the password, changes the hostname, SSH keys, and adds a public key.
Works great on 3B and 3B+ devices.
However, the 3A+ refuses to boot. It get's stuck on the rainbow screen.
No ACT light, so it looks like it's not even trying to run the boot sequence.
I can manually write the latest (2018-11-13) raspbian and it works fine, so it might just be that the PiBakery image is out of date with respect to 3A+.

Trying to work around it with a custom image, but having trouble making the required PiBakery edits to the base image.

Plasmarobo, 2019-03-08:
It does seem to be something wrong with the boot partition:
If I just copy the boot files from the latest raspbian-lite image, it seems to boot up.

renegadeandy, 2019-05-11:
I have hit this and believe I get the same outcome, @Plasmarobo any work around?

Plasmarobo, 2019-05-11:
I copied the start*.elf files from a known-good RPI 3+ image.
Worked like a charm. Just overwrite them in the boot partition.