davidferguson/pibakery #179: Is this project still alive?

pcs3rd, 2019-06-13:
Well...?

yombo, 2019-07-30:
It appears not. So sad, such a great tool.

dannystaple, 2019-07-30:
Perhaps the maintainer could setup a go-fund-me or patreon page for people to support this work. I think this is an awesome tool. I'd love to use it in more orionrobots projects. Getting V2 off the ground, and support for the latest images would be awesome.

davidferguson, 2019-08-10:
Hey everyone.

I think the term "long unplanned hiatus" applies to the release schedule and development of PiBakery.

Really sorry for the lack of development in this last year, however I'm hoping to change that now. See https://www.pibakery.org/future.html and also #180 for more info on this.

bd8392, 2019-10-26:
also love PiBakery and it would be great to see v2 =)

pcs3rd, 2019-10-27:
That's totally fine @davidferguson , until the update on here and twitter, it really wasn't clear what the status of the project was, thank you for such a cool tool!

bd8392, 2019-10-29:
@davidferguson I just tested the prerelease of 2.0 on windows with a pi 4 and buster...it sadly doesn't work...everything seems to be successfull, but when I try to boot the pi, it ends up in kernel-panic =(

Also you have to launch pibakery v 2.0 manually with administrator privileges...otherwise it gives a error message

ericraymond, 2019-11-18:
There are a few pull requests that are pending approval/merge.....

tvories, 2020-05-26:
@davidferguson would you consider moving pibakery to a github org or passing on ownership to someone who wants to maintain it?  I'd love to push some updates but there are no responses to PRs.

alexhunsley, 2021-12-20:
If the project is on haitus, can I suggest adding a message to the top of relevant pages saying "This project is not currently maintained and may not work" or similar? It might save others a little time. I was reading about it and got excited to try it, but the download links are all broken.

closed 2024-12-29 by pcs3rd