davidferguson/pibakery #180: Future of PiBakery

davidferguson, 2019-08-10:
I'd recommend having a read of [this page](https://www.pibakery.org/future.html) before this issue. That goes over more of the community side of the future of PiBakery, whereas this issue will focus more on the technical side of things.

---

Firstly I'd like to apologise to everyone about the lack of development on PiBakery. Through university work, other projects, (and just just pure laziness!) there hasn't really been much to show in this last year, and that's a real shame (and almost entirely my fault).

While I did re-write a lot of PiBakery for v2 - which as of yet is still the initial pre-release I published last year - there have been a number of changes/developments in the past year that make me want to re-write portions of it again. These are:

- **[Etcher SDK](https://github.com/balena-io-modules/etcher-sdk) has been released.** When I was writing v2, I was aware that the currently used module - [etcher-image-write](https://github.com/balena-io-modules/etcher-image-write) - was depreciated and not supported any more by Balena.io (formerly Resin.io). However at the time, no replacement was available, and I felt that using the depreciated package was better than using a combination of methods (including [my own fork of win32diskimager](https://github.com/davidferguson/CommandLineDiskImager), which I wrote without any C knowledge!) that was present in PiBakery v1.
However since then, [Etcher SDK](https://github.com/balena-io-modules/etcher-sdk) has been released and is in active development by Balena.io. This is what PiBakery should be using for writing SD cards, as it allows PiBakery to leverage the huge stability and regular updates/fixes of Etcher (which is a fantastic application!).

- **I've now used (albeit briefly) React.** If you've looked at the source of PiBakery (v1 and v2), you'll notice that the UI in v1 was very coupled to the logic, and in v2, while the separation is better, it's rather clunkily handled in pure JavaScript (not even jquery!), which makes it a bit of a pain sometimes. My defence of this is that vanilla JS was all I knew at the time, and it did work!
I'd rather like to re-write the UI in React (or something else, if people have other suggestions, or are familiar with other frameworks) to separate out the UI and logic. However, this obviously isn't a trivial change.

Also, one of the most requested features of PiBakery is to have a command line version, and currently, with the UI tied to the logic and disk-writing, this isn't possible without extensive work.

Additionally, PiBakery was my first large JavaScript project, and first outside of vanilla JavaScript in the browser. A lot of the things I did are old, now considered poor practise, or were poor form back when I did them (look no further than [this line of pure callback hell](https://github.com/davidferguson/pibakery/blob/ce81d939a9ded9e7fe115c54ddeb0b65cfe7c3c6/app/pibakery.js#L816) from v1!). I'd never used Electron before, and I'd never published a large project on GitHub before either. What you see in the code (especially in v1) is my experimentations and hacks to *just get it working*, not to produce a well written application, that others can contribute to.

---

Those are the main reasons I'd like to re-write a lot of PiBakery, however I'd love to hear what users and contributors think. Ideally, it would be nice to close a lot of the issues and PRs as well, as I've been very neglectful of them this last year.

Please comment below with your thoughts, and if you have experience with any of the technologies PiBakery uses or I'm proposing to use (Electron, Etcher SDK, React, etc.) please get in touch, as I want to do it right this time, and avoid the poorly written code that I initially wrote.

Thanks!
David

davidferguson, 2019-08-11:
Here's an outline of the features I'd like for PiBakery v3. This list may change over time:

- separate the core of PiBakery and the GUI into two separate packages, allowing for command line use
- use Electron's standard [auto updater](https://electronjs.org/docs/api/auto-updater), rather than no updater (v1), or a custom updater (v2). This requires the comparatively expensive code signing certificates, but I have enough in the donations pot to cover that for at least a year.
- use Etcher SDK rather than etcher-image-write
- use [Scratch Blocks](https://github.com/LLK/scratch-blocks) rather than Google Blockly
- use React for the UI
- expose the (currently hidden in v2) options that allow users to point PiBakery to use their own block repos, rather than the main PiBakery one

claremacrae, 2019-08-12:
Hello David, I'm a huge fan of PiBakery and would love to support this - and you - in any way I can.

Whilst I'm not (yet) familiar with these specific technologies, I am pretty experienced with projects that make incremental changes to existing code, to keep systems working and usable along the way.

I'm also experienced in getting hard-to-test projects under test...

So if I can help in any way, like with initial discussions about ways to approach this project - such as via Skype, Facetime or similar - or perhaps a remote-pairing/screen-sharing session to look at options, please do let me know - I'd love to help, and I love to learn - and from October onwards I have time!

henzard, 2019-08-15:
I suggest angular and I am happy to do the frontend and C#

solder-monkey, 2019-08-19:
In order to support CI/CD pipelines, it would be great if the command line CLI supported writing out to a IMG file rather than just to a physical SD card.

mr47, 2019-09-08:
@davidferguson I can help with it (React and others). Feel free to ask any React questions.

EternityForest, 2019-10-09:
I would suggest Vue.js instead of react. The template based syntax is cleaner than any framework I've ever seen and has almost no learning time (Although it's a little less well known).

Full rewrites are time consuming though.  Are you sure you can't just incrementally upgrade towards where you want to be?

RobertoC27, 2019-10-28:
I'd like to help with React as well (and with anything else I can), this seems too good of a project to just let it fade away.

DoctorDib, 2020-05-19:
Hi David, 

Just out of curiosity, is the PiBakery V3 project still on going, and does it have any expected beta release date?

It's just a great piece of software and would be a true shame to lose it!

James.