davidferguson/pibakery #181: Raspberry Pi 4 Support

henzard, 2019-08-14:
Hi I tried several solutions Pi 4 boots with buster standard but not if I use the same image with Pi Bakery.

rubger83, 2019-10-24:
I would also be very interested in a version of PiBakery which is supporting buster. Is there any chance, that it is available in the near future?
When I try to use an image of buster with PiBakery, I get message that the commands I use will not work with buster.

sanyer, 2019-11-15:
@rubger83 great, would you mind sharing details?

rubger83, 2019-11-18:
At first I had trouble with CR LF at the end of each line in the scripts (.py and .sh), because I downloaded the zip file of v2 and somehow my Windows changed the end of line signs (for use with Linux there has to be only LF). I had to download the Scripts using git clone.
V2 works well with buster, but I had trouble setting up the Wlan at firstboot. This helped: https://github.com/davidferguson/pibakery/issues/172
What kind of problem do you have? Have you checked the log files in /boot/PiBakery/ ?

tvories, 2019-11-20:
@rubger83 how did you fix the CR LF on windows?  I didn't see a solution that solved it.

rubger83, 2019-11-21:
First download the zip file of V2 from here
[https://github.com/davidferguson/pibakery/releases/tag/v2.0.0](https://github.com/davidferguson/pibakery/releases/tag/v2.0.0)
Unzip somewhere on your Windows machine.
As you have no control over what happens to the EOL when unzipping, you have to use git clone, to get the scripts in their original version.
At first you have to tell your git to leave the line endings in all files as they are with this command
`git config --global core.autocrlf false`
then clone 
`C:\>git clone https://github.com/davidferguson/pibakery.git C:\git`
then replace all .sh and .py files in the unzipped folder with the ones in your git folder
Then you can use the PiBakery.exe in the unzipped folder by starting as admin

tvories, 2019-11-22:
That fixed it!  I had to copy over `resources/app/pibakery-blocks` and `resources/app/pibakery-raspbian` and that seems to have fixed the issue of the kernel panic.

tbandtg, 2020-07-27:
@tvories  @rubger83 Thank you so much DenverCoder9