davidferguson/pibakery #182: V2 Compatibilty with Catalina

jessearmand, 2019-11-06:
I've downloaded the latest pibakery on latest stable version of Catalina.

I setup the basics, installed all the Raspbian Lite and Raspbian Full. Then, I tried to write Raspbian Full on a newly formatted microSD (FAT32) card with 32 GB size.

Then, it fails with an error:
[image]

Is it because of this change?

> PiBakery no longer bundles .img files in the installer/program. Instead, the user must supply their own Raspbian .img file. This means that any Raspbian-based .img can be used, with the possibility of other distros in the future

Then, should I put the images inside `/Library/Application Support/PiBakery/os`?

christophevg, 2019-11-24:
This seems to be related to stricter disk access rules in Catalina (found a reference for the dependency Etcher: https://github.com/balena-io/etcher/issues/2833)

A way around it is to run PiBakery as root:
```bash
sudo /Applications/PiBakery.app/Contents/MacOS/PiBakery
```

SergeStinckwich, 2019-12-15:
I got the same problem.

arturojain, 2021-12-22:
On `macOS Monterrey` this is not longer working, anyone has a fix?