davidferguson/pibakery #187: Pasting test in macOS won't work

besi, 2019-11-22:
Let's assume I want to execute a script on the first boot:

    sudo systemctl disable hciuart

I would simply want to copy and paste this into a script block. However, copy and pasting into the text fields won't work in macOS Catalina 10.15.1 (19B88).

As of now I have to type the command which is somewhat tedious and error-prone for longer commands.

JoschaMiddendorf, 2020-05-10:
I can confirm that. It would be great to be able to use copy and paste.

codygulley, 2020-05-24:
This almost makes this almost unusable.  In my case I want to curl a bunch of resources, but the URLs are so long.

ductape, 2020-06-17:
For the `Add public SSH key for authentication` block, the inability to copy & paste makes it impossible to use. The workaround I found is to use the [Pre-Release v2.0.0](https://github.com/davidferguson/pibakery/releases/tag/v2.0.0) and:
-  Add the _Add public SSH key for authentication_ block
-  Export the settings to .xml file
- Edit the XML in a text editor to actually paste in my public SSH key 
```
<block type="authorizekey" id="Qtz3:ho,racy{`v~9wNE">
                            <field name="1">PASTE SSH PUBLIC KEY MATTER HERE</field>
                            <field name="2">pi</field>
```
- Import the settings back into PiBakery. 

I could not get the export functionality of version v0.3.6 to work in macOS Catalina 10.15.5. v0.3.6 is the most recent released version to include macOS package and is still the latest version in [Homebrew](https://github.com/Homebrew/homebrew-cask/blob/master/Casks/pibakery.rb).

RobertStickney, 2020-09-15:
Another work around:
1. Clear text in text box.
2. Drag and drop text into the text box.