davidferguson/pibakery #189: Error in the running the program

luisfegomezPE, 2020-08-10:
[image]

I have follow all the steps to install pibakery, but in the last step I recived this message. How can I solve this?

thanks.

jmkebalo, 2021-05-04:
Hello,
I'm not mastering npm, but it seems the module semver was not install.
check on node_modules folder.
If any semserver, delete it, and run `npm install` again.

I had the issue with electron, it works for me

tbandtg, 2021-05-12:
Yeah deleting it did not seem to work. It is in there and gets replaced but I get an error when running.

jmkebalo, 2021-05-13:
I'm not sure to understand: If you delete it, where is it?
You should really remove manually the folder so it's not there anymore, and run the install again. It will download again, hopefully in a more proper way.

Can you share the error?

tbandtg, 2021-05-13:
[image]
Here is the error I get, I think it is downloading a new version of semver that is no longer compatable.

RPICtrl, 2021-05-13:
I Think it connects to pibakery.org and the site is Down.

jmkebalo, 2021-05-14:
There is an issue while providing a version to the semver module.
I'm so not an expert on this part... I just put a quick (ugly) fix by hardcoding a version on the file node_modules/semver/server.js :
[image]

It seems to work. But there is other issue to detect SD card... So I switch to Ansible to configure my rasp when the OS is installed.
Not a custom image but conf is deployed easily once the SSH keys are set.

cllackey, 2021-05-17:
I'm using PiBakery v2 and this is a quick fix I found to the SD card error. In the main PiBakery project folder, go to resources/app/main.js and comment out line 104 which has this:
`electron.app.dock.hide()`
I was able to write to my SD card with no errors and nothing wonky happened.