davidferguson/pibakery #197: SD Write Failed on macOS Monterrey

arturojain, 2021-12-22:
After upgrading to macOS Monterrey I can no longer finish writing on the SD Card. OS gets written OK, but fails to install any of the blocks I configure.

[image]

I'm running as sudo to overcome #182

Sam-M-Israel, 2021-12-23:
Not part of this issue but you are using a mac as well. I'm not able to download it from the site: https://pibakery.org/docs/install-mac.html. How did you manage to download it or are you building it from the source?

arturojain, 2021-12-23:
@Sam-M-Israel I've been saving a copy. I can send it later. I'll edit this comment or create a new one.